MNieddu91/Marlin-AI3M-BLTouch #10: Unable to export compiled binary through Arduino 1.8.9!

jeffgram, 2019-04-13:
Hello My professional dear,
I am having Anycubic Mega S. 
I am currently having a new issue to export the compiled binary as hex through Arduino 1.8.9 software, which I never had this issue before. But now I have encountered this unexpected error of red highlight at the bottom saying "Sketch uses 125320 bytes (49%) of program storage space. Maximum is 253952 bytes.
Global variables use 6638 bytes (81%) of dynamic memory, leaving 1554 bytes for local variables. Maximum is 8192 bytes.
Low memory available, stability problems may occur."
Could anyone of you help me out? I would owe you my appreciation and support.
Here is my email: sawak.the.great@gmail.com
Best regards,
Jeff.

MNieddu91, 2019-04-15:
1554bytes is more than enough. Just ignore this warning and flash the hex. Marlin need around 1000bytes to run properly.

jeffgram, 2019-04-16:
> Hello,
> Thanks for your reply. I do really appreciate.  Could you please help me
> out.I have unexpected issue since yesterday and I didn't sleep at all until
> morning. I have AnyCubic i3 Mega upgraded to Mega S. Yesterday, after i
> flashed Marlin 1.1.9 on my 3D printer. I printed many files and it was
> printing without filament out of the nozzle. I checked the extruder, which
> was still rotating.Then, i realized that slicer had an issue of its
> configuration profile. I checked it was all ok as I used to print through.
> Finally, i thought its extruder had an issue. Therefore, i replaced it with
> my old extruder Mega i3 and flashed it, yet result showed the same. I did
> all i could until I went so disappointed and frustrated.
> If you could help me to solve my issue, i would make a donation for you.
> This is my week old 3d printer.
>
> Best regards,
> Jeff

Hello,
Thanks for your reply. I do really appreciate.  Could you please help me
out.I have unexpected issue since yesterday and I didn't sleep at all until
morning. I have AnyCubic i3 Mega upgraded to Mega S. Yesterday, after i
flashed Marlin 1.1.9 on my 3D printer. I printed many files and it was
printing without filament out of the nozzle. I checked the extruder, which
was still rotating.Then, i realized that slicer had an issue of its
configuration profile. I checked it was all ok as I used to print through.
Finally, i thought its extruder had an issue. Therefore, i replaced it with
my old extruder Mega i3 and flashed it, yet result showed the same. I did
all i could until I went so disappointed and frustrated.
If you could help me to solve my issue, i would make a donation for you.
This is my week old 3d printer.

Best regards,
Jeff

closed 2019-04-17 by MNieddu91