MNieddu91/Marlin-AI3M-BLTouch #11: [BUG] Printer doesn't detect BLTouch trigger

mLupine, 2019-05-08:
At this point, I'm basically out of ideas on what could be causing my issues. I've installed a BLTouch clone (3D Touch) on my Anycubic i3 Mega, and as far as pushing the pin up and down, the probe works fine. However, it doesn't seem to react to triggering the pin when it needs to. I first tried the procedure from the Wiki, but pushing the pin while G29 was in progress did not result in the carriage moving to the next point – it tried to go further down. I then tried a M119 command which, correctly, returned the "open" state. After disconnecting the probe, M119 returns "triggered" which also seems to be correct. However, the result of the "M43 S" test that I did clearly tells that the printer does not receive the trigger notification from the BLTouch, and I have no idea what could be the issue.

Attaching a screenshot of the M43 S command result:

[image]

**Printer:**
 - Firmware version: v1.1.9 (latest commit)
 - Printing method: OctoPrint
 - Model: i3 Mega

Any help will be greatly appreciated!

Thanks,
M.

MaxBeliar, 2019-05-13:
I've faced the same problem. So, I am also need help.

dleask, 2019-05-13:
I had the same problem. It meant the servo connection was OK but the black and white sensor wires were reversed. 

Regards

Dave

MaxBeliar, 2019-05-13:
> I had the same problem. It meant the servo connection was OK but the black and white sensor wires were reversed. Regards Dave
> […](#)

Unfortunately, this doesn't work for me.

dleask, 2019-05-13:
Then I would suggest you may have a connectivity problem as the sensor pins on the motherboard are not detecting a closed signal when the probe pin retracts. 

Regards

Dave

MNieddu91, 2019-05-25:
Problem ist false or defective wiring of Z Endstops (white and black wire)

mLupine, 2019-05-30:
I managed to resolve the issue, and the issue was - indeed - wrong wiring. My BLTouch unit had cables connected to the plug in a wrong color order – I just had to reverse the white and black wires and got everything to work. Thanks for the help!

closed 2019-05-30 by mLupine