MNieddu91/Marlin-AI3M-BLTouch #12: [BUG] Auto-leveling stopping during process

jultur, 2019-05-27:
Hi, 
I was following this tutorial to install a 3D Touch (BLTouch Clone) probe on my anycubic mega i3. After correcting some problems due to the wiring, I managed to get the probe working with the gcode commands like : extanded pin M280 P0 S10, retracted pin M280 P0 S9 et with the commande M43 S that tests the probe i manage to get it triggered. 
However, when i start the auto-leveling process with the menu or with G29 , the printer rarely finish the process.
On top of that the process seems to got further when the heated plate is cold. I never manage to get all points of the auto-leveling when the heated bed was hot (60°). 
I also try to start printing with the auto-leveling on a cold plate but it looked like the working space was move of a 110 offset. Indeed the home of the printer was in the middle of the X axis. the X carriage was getting stuck to the side where there is no end stop and it was impossible to move towards the X end stop when the X carriage was in the middle. 

Here is a video of the printer stopping during the process:

[https://www.youtube.com/watch?v=ha9RR8GpPkM&t=1s](url)

I don't really see where the problem is coming from and I could really appreciate a hand. 
Thanks !

ThronX, 2019-05-29:
I have the exact same problem.
I tried a different sensor (bltouch instead of 3d touch) but with the same result.
Maybe somebody could help to solve the problem. I don't think it's a fw bug. I think it kind of a configuration issue.
Sometimes I could finish the autoleveling process. But mostly it canceled during the first an 4th measuring point.

MNieddu91, 2019-06-04:
Do you get an error message? Had this issue in the beginnings with BLTouch. 
Try to higher the bed or for testing put one, two, three or maybe more A4 papers on the bed. If this will work, you need to put a washer between mount an BLTouch.

jultur, 2019-06-05:
OK I also bought the bltouch, add some more washer between the mount and the BLtouch, higher the bed and changed the  #define Z_PROBE_LOW_POINT -5 .... nothing is working. I'm still seeing the same behaviorwith the messages :
Error:Probing failed
[ERROR] Error:Probing failed

ThronX, 2019-06-05:
I think I found a solution for myself.
I tried to use the Marlin 2.x for the Anycubic. 
I configured everything to use the bltouch. It seems that it's working.
I'm using it for three days now with no problems.
The only thing which you could not use is the special menu. You have to configure it with commands in the first place. Afterwards you could setup your start script.

jultur, 2019-06-05:
> I think I found a solution for myself.
> I tried to use the Marlin 2.x for the Anycubic.
> I configured everything to use the bltouch. It seems that it's working.
> I'm using it for three days now with no [...]

Is it possible to get your files ?
here is my mail : jul.turpin.jt@gmail.com

ThronX, 2019-06-05:
I'll post it here in case somebody needs it.

It's edited from this repo from davidramiro.
[https://github.com/davidramiro/Marlin-Ai3M-2.0.x](url)

[Marlin-Ai3M-2.0.x-bugfix-2.0.x-with-bltouch.zip](https://github.com/MNieddu91/Marlin-AI3M-BLTouch/files/3258181/Marlin-Ai3M-2.0.x-bugfix-2.0.x-with-bltouch.zip)

MNieddu91, 2019-06-07:
Do you talk about the problem in my bugfix 2.0.x or in marlin 1.1.9 ? 
In Marlin 2.0.x it is a know bug, i have the same issue. That's why it's called bugfix.

cpm64, 2019-07-22:
It seems it doesn't work for me. It is stopping randomly above one of these positions of the 5x5 matrix: 3, 8, 13, 18 or 23, very very rarely in other points. So , most of the time is on the middle of X.
The head position above the point, the probe is extended (blue LED), Z starts to move down but after a few millimeters it stops and is raised back to the initial Z position above the point. The probe is still extended and blue LED still on. The probe nor the head is reaching the bed. OctoPi is reporting 'probe error' and M119 reports everything as OPEN. The mechanical Z stop is not reached even if the screws are very close to the switch.
The probe seems to be OK as it passes all the tests (M280 Px Sy) but it cannot perform a reliable G29.

- I tried with other shielded cables , re routed to not touch the rest of the wires. I eliminated the bad wiring assumption.
- I also disabled fans and heaters when probing
- Increased delay to 1000ms
- Increased Z_PROBE_LOW_POINT to -100.
- Adjusted the mechanical limit switches to allow the head to be below the bed thus eliminating mechanical stops triggering.
- The Z-Offset is -2.80

My setup:
- Latest 1.1.9, cloned on July, 15th  https://github.com/MNieddu91/Marlin-AI3M-BLTouch
- TMC2208 , TMC2208 UART on E0 with TX on D14 and RX on D15
- Original BLTouch V3.0 (enabled all the mandatory directives)

cpm64, 2019-07-22:
It seems to work with Marlin 2.0 at least I have a first print at the proper level, sticking to the bed and I can see the Z rods moving when traveling XY at a specific layer height.

cpm64, 2019-07-24:
I restarted 1.1.9 config from scratch and with configuration below I passed 5 G29 without issues. On some points the probe do a pin on/off for 2 times before Z start to move down but is doing the work and is not reporting "probe error" anymore:
- move the software serial wires of E0 TMC2208 UART from D15 and D14 to TX-D20 and RX-D12 (it seems D15/14 are used by TFT as well). Not sure this is related to BLT issue

- performed these changes, only changes that I consider impacting the BLT are listed below:
`//#define SHOW_BOOTSCREEN
#define E0_DRIVER_TYPE TMC2208 // pins configured in pins_TRIGORILLA14.h D20 TX and D12 RX
#define Z_MIN_PROBE_ENDSTOP // Use the default D2 pin defined in pins_TRIGORILLA_14.h see the code changes done in that file
 #define BLTOUCH
 #if ENABLED(BLTOUCH)
  #define BLTOUCH_DELAY 1000   // 100 is too small and either skips measuring (Marlin 2.0) either stop above the measuring point in Marlin 1.1.9 . 
 
  #define BLTOUCH_V3
   #if ENABLED(BLTOUCH_V3)
    #define BLTOUCH_FORCE_5V_MODE // Trigorilla has 5V logic
    #endif
 #endif

 #define MIN_PROBE_EDGE 10

#define XY_PROBE_SPEED 8000 // Increased to 8000 as defined in Marlin 2.0

#define Z_CLEARANCE_DEPLOY_PROBE   15 // Z Clearance for Deploy/Stow // https://www.antclabs.com/manual
#define Z_CLEARANCE_BETWEEN_PROBES 10 // Z Clearance between probe points // https://www.antclabs.com/manual
#define Z_PROBE_LOW_POINT          -2 // Farthest distance below the trigger-point to go before stopping
 
 #define LEFT_PROBE_BED_POSITION 10
 #define RIGHT_PROBE_BED_POSITION 200
 #define FRONT_PROBE_BED_POSITION 10
 #define BACK_PROBE_BED_POSITION 200
 #define MIN_PROBE_EDGE 10
#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command // https://www.antclabs.com/manual
#define TRIGORILLA_VERSION 1  // Trigorilla 1.1
`

Not sure if is relevant: during this test the Z mechanical end stops are triggered under the bed level about -2mm, yeah the nozzle push down the bed on G28.

Hostiss, 2019-08-27:
I can confirm , i had exactly same problems, with  a genuine BL Touch (version 3.1), as described in this thread. Doing G29 was very unreliable, sometimes it does work, but mostly, it was crashing in the middle of probing or results were very unreliable, because pin was extending, immediately after  retracting then moving to the next spot. I was freaking out thinking, that there is something wrong with the probe, since it's acting exactly the same, as cheap clones. 

I made same changes to config file, as @cpm64 described above (except these ones related to TMC2208), and i can confirm, it's working much better now (1.9.1 Marlin).

Right now, even, if it's still acting the same (extending and retracting almost instantly),it still can continue probing, because this doesn't count for measurements, so thank you very much, for your post.

I have Anycubic I3 Mega S with Trigorilla 1.1 and TMC2208 stepper motor drivers (with no reversed connections)

cpm64, 2019-08-27:
Great if is working for you. However after i did the changes posted in my previous comment above i still have a failure rate of 5 to 10 percent, thus i switched to MNnieddu91's Marlin 2.0.x and never looked back as is working flawlessly. So far, from my research the key is the BLT3.x  hw changes and the BLTOUCH_DELAY . I will also have a look on the changes btw TMC2208 lib used in 1.1.9 and the new TMC LIB used in 2.x as I suspect a possible side effect of the EMI in 1.1.9 - just a thought, may be is nothing related to this.

The only annoying thing is that G29 will take about 3- 5 min to complete , using it on every small print is not an option for me. Here the erprom mesh is handy.

closed 2020-04-07 by MNieddu91

stef-ladefense, 2020-04-12:
Sorry to reopen,
I had the same problem, and I ended up with the oscilloscope probe in my hands to discover a lot of noise everywhere, the wires play the antennas!
I started playing with the ENDSTOP_NOISE_FILTER which solved the concerns, but which also added a small error.
so I took out the soldering iron and solved the problem with the installation of 100nf ceramic capacitors on all the endstops of the machine (x4).
Anycubic unfortunately did not consider it useful to include this capacitor on its endstop.