MNieddu91/Marlin-AI3M-BLTouch #13: error "failed to enable bed leveling"

jimbo2112, 2019-06-19:
I have followed the instructions for calibration and run into an issue.  figured out the Z offset to be -1.6. I entered this into my firmware, compiled, and reflashed. I checked the location of the nozzle using G1 Z0 and a 0.13mm feeler gauge. I can just barely slide the gauge between the nozzle and the bed. The 0.15 gauge will not fit. The 0.1 has a visible gap.

After this I went to cura and changed the script per the directions. I added G29 after G28.

The first time it homed, then ran g29, then had an error of "failed to enable bed leveling"
I thought maybe it was the M502 and M420 codes from the mesh level. I deleted those 2 and the same problem occurred.
Then I tried removing the G29, adding the M502 and M420 codes back.
I used the special menu to select "Auto bed leveling" It ran with no problems. 
I saved the EEPROM and rebooted.

This time the printer ran but I had massive failure right away with bed adhesion and the skirt. I could tell from the get go the nozzle was not close enough. The filament is not squished, rather it was nearly round. As it is printing the distance between the nozzle and the bed appears to be about 1.5 to 2 mm, not the 0.13 or so it should be. 

Below is the starting script with the g29 NOT added.
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 Z0 ;move X/Y to min endstops
M501
M420 S1
G0 Z0.15
G92 E0 ;zero the extruded length
G1 X40 E25 F500 ; Extrude 25mm of filament in a 4cm line. Reduce speed (F) if you have a nozzle smaller than 0.4mm!
G92 E0 ;zero the extruded length again
G1 E-1 F500 ; Retract a little
G1 X80 F4000 ; Quickly wipe away from the filament line
M117 Printing…
G5

When I entered the G29 command after the G28 I tired with both with and without the M502 and M420 commands.  I did not delete any other command.

cpm64, 2019-07-22:
Same problem here, in the rarely times when the BLT is working (complete a G29 without freezing above probing points 5x5 matrix, issue reported here: https://github.com/MNieddu91/Marlin-AI3M-BLTouch/issues/12#issuecomment-513680694) I cannot enable the mesh, 

I followed and logged terminal output for every command described in the guide (https://github.com/MNieddu91/Marlin-AI3M-BLTouch/wiki/Adjust-configuration)

G28 ; (homing should be done as usual)
G1 Z20 ; (Raise Z axis)
G1 X110 Y110 Z100 ; (center print head and position at height Z=100)
G29 ;  (axis moves to the first measuring point (X10 Y10) , the pin is extended and the print head moves down)
`
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +2.442 +2.550 +2.645 +2.692 +2.690
Recv:  1 +2.420 +2.532 +2.650 +2.687 +2.682
Recv:  2 +2.417 +2.527 +2.625 +2.662 +2.657
Recv:  3 +2.445 +2.535 +2.615 +2.635 +2.607
Recv:  4 +2.412 +2.490 +2.562 +2.560 +2.542
Recv: 
Recv: X:190.00 Y:225.00 Z:12.46 E:0.00 Count X:15200 Y:18000 Z:6000
Recv: ok
[...]
Send: M851
Recv: echo:Probe Z Offset: 0.00"`

Perform a G29 again, results:
G29
`Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +2.442 +2.550 +2.648 +2.692 +2.695
Recv:  1 +2.422 +2.535 +2.630 +2.690 +2.690
Recv:  2 +2.417 +2.523 +2.617 +2.662 +2.660
Recv:  3 +2.437 +2.532 +2.615 +2.640 +2.620
Recv:  4 +2.407 +2.487 +2.560 +2.557 +2.545
Recv: 
Recv: X:190.00 Y:225.00 Z:12.46 E:0.00 Count X:15200 Y:18000 Z:6000
Recv: ok`

M420 S1 ; Bed leveling ON
`Send: M420 S1Recv: echo:Bed Leveling OnRecv: echo:Fade Height OffRecv: ok`

**Now, the Z-offset calculation**:

M211 S0 ; Disable software endstops
G28 ; (homing should be done as usual)
G1 Z5 ; Lift Z 5mm
G29 ; (Approach measuring points with BLTouch)
`Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +2.422 +2.542 +2.628 +2.687 +2.692
Recv:  1 +2.412 +2.517 +2.625 +2.667 +2.675
Recv:  2 +2.410 +2.505 +2.612 +2.657 +2.650
Recv:  3 +2.427 +2.515 +2.605 +2.625 +2.600
Recv:  4 +2.380 +2.455 +2.537 +2.545 +2.530
Recv: 
Recv: X:190.00 Y:225.00 Z:12.47 E:0.00 Count X:15200 Y:18000 Z:6000
Recv: ok`

G1 X110 Y110 Z5 ; (Move print head back to center)
G1 Z0G1 Z-0.20...G1 Z-2.50 ; Found it on -2.50

M114 ; Read current positions
`Send: M114
Recv: X:110.00 Y:110.00 Z:-2.50 E:0.00 Count X:8800 Y:8800 Z:41
Recv: ok`

M851 Z-2.50
M500 ; Store settings in EEPROM
G28 ; (homing should be done as usual)
G29 ; Execute ABL

`Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 -0.072 +0.040 +0.125 +0.175 +0.187
Recv:  1 -0.088 +0.015 +0.125 +0.170 +0.177
Recv:  2 -0.092 +0.010 +0.112 +0.158 +0.155
Recv:  3 -0.072 +0.017 +0.095 +0.125 +0.102
Recv:  4 -0.122 -0.043 +0.037 +0.040 +0.033
Recv: 
Recv: X:190.00 Y:225.00 Z:17.47 E:0.00 Count X:15200 Y:18000 Z:7000
Recv: ok`

G1 X110 Y110 Z5 ; (Move print head back to center)
G1 Z1 ; Slow steps, use multiple commands till Z0, this avoid crashing the head to the bed) - The Z0 is correct, the paper is like I set it above.

From this point updated the FW and included #define Z_PROBE_OFFSET_FROM_EXTRUDER -2.50 , reflashed, 'M502, M500', I run the LCD Auto leveling then in Cura just underneath G28 Z0 I added:
'M501
M420 S1'

Tried to print a cube but the head is far above the bed and testing manual X movement it doesn't show me any Z-axis rods movement.

I tried with both the mechanical endstops above the bed level and under the bed level (the head is pushing down the bed by 1-2 mm).

It seems that ABL is ignored.

cpm64, 2019-07-22:
After switching to Marlin 2.0 , G29 seems to work and after retrying all the above steps everything is (hopeful) fine. I can even see the Z rods moving when traveling XY on the same layer height.

jimbo2112, 2019-07-23:
" After switching to Marlin 2.0 "
I thought it was recommended to stay on 1.1.9 with the I3 Mega.  Does 2.0 support mesh leveling?

cpm64, 2019-07-23:
Considering that:
- I have the Z stop mechanical endswitches about 3mm under the bed level (this is not really ok in case of troubles it will crash into the bed, I will adjust the endstops to trigger when nozzle is barely pushing the bed);
- G28 , when homing the nozzle is pushing down the bed for about 3mm
- with ABL activated the paper drag is the same on corners like it is in the middle of the bed. The bed mesh shows ~0 in the center and about ~ -1.5 on corners
- I have printed a test cube right on the X0 Y0 and is OK

I suppose is working. I did it once last night after 5 days in which I tried all the possible settings in 1.1.9 without any luck, I admit I was tired and may be I was wrong - however the cube is OK :-), I can see the Ultrabase dots pattern on the bottom of the cube , no under no over extrusion, no squashed filament, very firm bound between layers. I will redo this evening and try to take some references on the rods (paint marks) to see if is really adjusting when moving X carriage.

By the way, on real Z0 (no ABL, mechanical switch triggered) is your nozzle pushing the bed down?

closed 2020-04-07 by MNieddu91