MNieddu91/Marlin-AI3M-BLTouch #16: [FR] Mega-S Support

b-reich, 2019-09-28:
Please add support for the Mega-S upgrade kit.

The updated extrueder allows printing with tpu. The firmware from davidramiro supports the mega-s.

davidramiro, 2019-09-28:
This firmware works fine with the Mega-S, just as mine does. The config of this repository even has the required e-steps set by default.

Psychokiller1888, 2019-12-06:
Hey! Thanks for the work! Just to let you know, but on a default Mega S, all the axis are inverted

agroezinger, 2020-01-15:
If you are using the stock drivers then yes, all axes are inverted. You have to change the stepper drivers in the configuration.h and recompile:

#define X_DRIVER_TYPE  TMC2208_STANDALONE // comment out for stock drivers
#define Y_DRIVER_TYPE  TMC2208_STANDALONE // comment out for stock drivers
#define Z_DRIVER_TYPE  TMC2208_STANDALONE // comment out for stock drivers
#define X2_DRIVER_TYPE TMC2208_STANDALONE
#define Y2_DRIVER_TYPE TMC2208_STANDALONE
#define Z2_DRIVER_TYPE TMC2208_STANDALONE // comment out for stock drivers
#define E0_DRIVER_TYPE TMC2208_STANDALONE // comment out for stock drivers
#define E1_DRIVER_TYPE TMC2208_STANDALONE // comment out for stock drivers
#define E2_DRIVER_TYPE TMC2208_STANDALONE
#define E3_DRIVER_TYPE TMC2208_STANDALONE
#define E4_DRIVER_TYPE TMC2208_STANDALONE

closed 2020-04-07 by MNieddu91