MNieddu91/Marlin-AI3M-BLTouch #18: [FR]BL Touch and Z-Endstops

Rich990, 2019-11-16:
hi, I am new here and just have a question but couldn’t find the right spot or button to post it, so I did it this way. With your latest version of Firmware WITH the use of BL Touch, what does the anycubic i3 Mega S uses for Homing? Does it used the Z-Endstops for that and the Bl Touch just for the ABL? 
Thanks for your help and answer!

RichardGottschalk, 2019-11-21:
I just successfully installed the BL Touch with the steps described in [wiki/Installation-BLTouch](https://github.com/MNieddu91/Marlin-AI3M-BLTouch/wiki/Installation-BLTouch).
The Anycubic i3 Mega is using individual Z min endstops for both Z axis (left and right one) to synchronize them.
With this firmware, both Z axis endstops are still working properly. By connecting the white/black pin to the X-carriage PCB as shown in the linked image, you are connecting it to a X-max endstop pin on your mainboard which you configure to act as your BL-Touch probe.

So, you are still using both Z-axis min endstops AND the BL Touch.
The Mega-S should work the same way, but may require some minor config changes for the extruder. Haven't updated my non-S, yet :D

Note: I am using a BL-Touch clone, which has its pins swapped on the 3-pin connector. Watch your probe's manual and swap them if needed.

Rich990, 2019-11-22:
Thank you very much!

closed 2019-11-22 by Rich990