MNieddu91/Marlin-AI3M-BLTouch #19: [BUG] - Auto bed leveling failure

shaddow501, 2019-11-17:


Rich990, 2019-12-07:
Have u fixed it? I have the same issue as you... even in the same directions..

shaddow501, 2019-12-07:
No it wasnt the TMC2208, back to stock drivers and it is still behaves the same.

I didnt succeed with that and no one has given me an answer, if it was something mechanically it should have been shown the problem everywhere, but the problem is consist on one side, same side every time.
Can I ask you when did you buy your printer? mine was bought about 2 years ago... I am trying to find a clue maybe it is related to the older printers and with the newer ones there aren't any issue with that, (even though it seems to me very strange since it is the same board and same firmware)
I dont see any answers from others, hey guys, did you try it how does it work with your printer, no one answer, even not the one that put this github, hey "MNieddu91" does it work for you or you see the same problem we see?
NO ANSWER FROM ANYONE!

dleask, 2019-12-07:
I had a similar problem on an Anycubic Mega-S. It turns out the two rods of the x axis were not in perfect alignment. The top rod was slightly forward on the right side by a fraction of a mm. As the BLTouch pin is forward of the nozzle that meant the Z offset at the left side was -1.1mm but at the right side it was -0.9mm. Unfortunately as auto level and single offset assumes the offset is constant you will find following the auto level that one side will be noticeably tighter on the right side. 
The solution is to micro adjust the alignment of the rails so that the offset is consistent at both ends of the x axis.

shaddow501, 2019-12-07:
dleask 
I dont quite understand  the X are the rods that the print head sitting on it (left and right movement) 
How it is it related if one rod is about 1mm longer than the other one?
Those are not curved rods they are straight, I can install 500MM rod that will be sticking out from each side will that solve my issue? or I didnt understand you well?

Lest say that the rods are curved a bit the bltouch should correct those curves since one side it lower and the other is higher..

so please explain yourself.

davidramiro, 2019-12-07:
>How it is it related if one rod is about 1mm longer than the other one?

It's not that they are longer, but that they are poorly aligned. I experienced the same on mine.

>Those are not curved rods they are straight

Oh no, they certainly aren't. This is a cheap printer and you get parts with bad tolerances. Swap them out for some Misumi rods + bearings and you will see what I mean. Also, don't get me started on the two rods of the Y axis! Had a lot of fun fixing them to be somewhat parallel.

How this would affect BLtouch, I cannot comment on, don't own that thing. Sorry.

Just as a general advice, it's a good idea to try fixing those irregularities. Sensors and software correction can only do so much and it might not work reliably if the whole carriage is bent in a slight way - especially since the BLtouch is fairly sensitive and precise.

Also: 

>NO ANSWER FROM ANYONE!
>I know, when I make something for others I usually respond to people, this issue was created 20 days ago and still MNieddu91 didnt have the time to respond if he see the same issue or what is different with his bltouch.

Yes, that's true, he didn't have the time to respond as he has obligations in the real world. Both of us simply can't put as much time into this project as we were before, sadly.I am sure there are other communities that might be able to help. Especially since MNieddu91 didn't write the Marlin BLtouch code himself, this is just a fork - you might be more successful looking through the issue section on Marlin's official GitHub as well as communities like RepRap Forums.

shaddow501, 2019-12-07:
Dear davidramiro, 

I have better results and better printing quality when I do not use the auto bed leveling option but level the bed mechanically, I have very clean prints. the problem is that I need to do it often and I wanted something automatically.  

As the answering time or even some kind of a short comment would be nice.
I didnt expect him to work on his modified code, just asked 2 things,
1. can you confirm that you dont have the issue we are having? (yes / no question)
2. what is your setup.

It doent take much time to answer does it?
If you purplish a code that you made changes in it, you are somehow obligated to that code, what have you done? what did you change so I do not fully agree with you david.

davidramiro, 2019-12-07:
>If you purplish a code that you made changes in it, you are somehow obligated to that code, what have you done? what did you change so I do not fully agree with you David.

He didn't write the code himself. That credit still goes to Scott Lahteine and all the other bright minds. He tweaked the config of this fork of Marlin for BLtouch and wrote a comprehensive guide. He has no obligation whatsoever. It's not like we get paid by Anycubic to provide you an updated firmware of their printer (without hefty security flaws - that should be their job in the first place).

It just gets exhausting. I'm trying my best to answer everything (I'd like you to take a look at the firmware's Thingiverse page which is at over 1000 comments as of now), many of the issues I'm responding to are literally mentioned in the description or my Wiki or at least spending 10 minutes on a search engine of their choice - people just tend to not be thorough. Also, you wouldn't imagine how many weird BLtouch issues got reported to MNieddu91 and me that in the end were just caused by cheap knockoff probes being used. (General advice: If it says "genuine" on AliExpress, it probably isn't).

Please don't take offense at my little rant, all I'm really saying is you will probably get help faster if you try asking other communities as well as this particular repo just uses a modified config. I'd love to help you but I do not have enough insight into BLtouch.

shaddow501, 2019-12-07:
Dear David

I am sorry but I didnt complain to you... you have decided to step in for "MNieddu91" you dont need to answer, and basically we are arguing about things that doesnt forward us  to solve this issue.

If you have some guidence or technical answers or willing to help to solve this issue, cool,

but at the moment by your answers we do not accomplish anything, are you his lawyer? 
I expressed here part of my issue some disappointment that the person that "tweaked" this code have not yet answered for 20 days.

You stepped in? why? do you have a solution or just into wasting everyone time?
Yours and the other that read this post and waiting for a solution.

dleask, 2019-12-07:
It’s not the length of the rods. If you look top down on the printer the rods should be aligned above each other. The scenario I’m highlighting is the mounting for the rod can be slightly adjusted. I noticed that the top rod at the right side was half a mm forward of the lower one. The effect of that was to tip the head forward slightly compared to the left side. I loosened the rod mounting bracket on the right, pushed it back within the limits of the bolt holes and tightened it back up. Now when the head moves from left to right, the offset between the nozzle and BLTouch remains constant. The auto level accurately maps the bed and the corrections applied now keeps the nozzle at a consistent height across the whole bed.

shaddow501, 2019-12-07:
Dear dleask, I will check that but as I understand of the auto bed leveling it should compensate for those deviations,

davidramiro, 2019-12-07:
>Further more, when I have post this question under your github code asking for maybe help from you, you have deleted my question.

Yes, because it's not related to my firmware. We had multiple interactions already (the latest one only a few days ago on the 2.0 Marlin) which of course I did not delete.

>but at the moment by your answers we do not accomplish anything, are you his lawyer?
I expressed here part of my issue some disappointment that the person that "tweaked" this code have not yet answered for 20 days.

Just wanted to give you a heads-up that he probably won't respond in a very timely matter due to real world obligations. Nothing more. Also, my first response was just to confirm what dleask mentioned, as it was the same on my machine. I just can't stand Caps Lock, which is why I commented on that part.

But you're right, this is getting way off topic. Sorry for that.

To get on topic, maybe I can forward your issues to him personally: Which kind of BLtouch are you using? v3? Also, where did you purchase it? Sadly there are many vendors that are impersonating Antclabs and sending out counterfeit probes.

dleask, 2019-12-07:
The bed auto levelling compensation does work but only when the x axis is perfectly aligned. It’s doing it’s job on your machine. You have a -2.7mm z offset in the middle of the bed. 
To test my theory go through the recalibration setting M851 Z0.0, Home the axis with G28 then run the auto level with G29. Next instead of moving to X110 Y110 to test the offset, do it at X10 Y110 and then at X210 Y110. I suspect you will find a difference of between 0.1-0.2 between the two readings. If that is the case, adjust the top rod until you get the same offset reading at both sides of the bed.

shaddow501, 2019-12-07:
Dear davidramiro

I first see in one of the links that someone suggested this one:
[image]

It is with the plastic pin (not metal)
Then I have found out that this is a clone, but already bought it.

The readings are very consists so I dont think the sensor is the problem, since it always the left side that us up and the right side that is down. 
If the sensor have a problem the bed leveling would have been different each time.
So for your question no it is not V3. (the China guys didnt yet clone V3 lol)

shaddow501, 2019-12-07:
Dear dleask
I have -2.7mm z offset since it is the recommendation of the installation of the sensor.
it should be about 2.5 +- 0.5  above the height of the nozzle. (so for me it is 2.7)

[image]

Again the auto leveling should correct it, if your printer is 100% stright you do not need the auto bed leveling.
if the sensor "sees" that one side is up constantly and one side is down constantly the auto bed leveling should  correct it. it doesnt.

further more if i disable the auto bed leveling I get very nice prints.
so again it is something else.

dleask, 2019-12-07:
Last attempt. 
Your x axis rods are twisted. As the head moves to the right, the head tips forward and lowers the probe relative to the nozzle. 
Your bed, like mine isn’t warped so when you manually level each corner with a piece of paper and run a print without auto levelling on you get a good print. 
Then you probe with G29, turn on auto levelling and the next print has one side of the bed scraping the nozzle when you print because the auto levelling applies compensation as it thinks the right side of the bed is lower than it actually is.

shaddow501, 2019-12-07:
The Probe an the nozzle are connected to the same surface so both are lowed the same.

can you provide a picture ?

dleask, 2019-12-07:
I have an Anycubic Mega-S with TMC2208 drivers on all axis. I started running David Ramiro's 1.1.9 firmware very successfully then added Marco's version with BLTouch support and fitted a genuine BLTouch (which does have a plastic pin). My bed is warped in the centre with a pronounced hump running down the centre of the Y axis. Not enough to ruin a print but enough for my critial eye to notice which is why I added the probe.

Unfortunately I'm not disturbing a well running printer to explain a point but I will share annotated photos which I have just taken. 

**There is a mechanical principle I'm trying to explain regarding the alignment of the X axis rods. If you do not understand or agree with the principle I'm explaining then I cannot help you any further as you will be fighing against the correct operation of the BLTouch and auto leveling feature.**

My printer is based on the Prusa i3 design with two parallel rods supporting the X axis. The print nozzle is around 25mm in front of the bottom rod, the BLTouch probe is another 25mm in front of the nozzle.  

[image]
Using your measurement as an example. When the print head is at the left of the X axis, the top and bottom rods are perfectly aligned on top of each other. If I drew a line from between the centres of the rods it would be perfectly vertical. in that case the BLTouch probe trigger point is 2.7mm below the print head.
[image]
At the right of the X axis, the rods are out of alignment. A line between the centres of the rods at the right mounting points when viewed from the centre of the printer is not vertical and leans towards the front of the printer. This has the effect of lowering the BLTouch tip/trigger point below the previous position, now greater than 2.7mm.

If you do a G29 at this point, the probe measurements will give an accurate profile of the bed, remember the probe measures deltas not absolute. Lets assume you nailed it, the bed is perfectly flat and you use some fine adjustments on the right side as it looks a bit high on the probe results. The probe readings are now level but when you print with auto levelling the nozzle is scraping the bed on the right. You manually drop the right corners a 1/4 turn, re run G29, try a print and it does the same.
The autolevel corrects for the lowering of the right side but cannot correct for the change in offset between the BLTouch and the nozzle as it has no way of knowing about that.

I'd start from scratch with the BLTouch callibration and do it twice. We know in the middle of the X axis you get an offset of -2.7mm (M851 Z0 and you need to drop the Z level 2.7mm to get the nozzle to lightly drag the paper used for levelling). First use X10 Y110 Z5 and measure it, next reset and use X210 Y110 Z5 and measure from there. I suspect you will find there is 0.2mm difference between the two sides.

If there is, use the picture below to correct a lower right side. Loosen the bolts marked in red, to raise the right side push the top rod back within the gap (1) and tighted the bolts and pull the bottom rod forwards and tighten the bolts and repeat the callibration. If the Z offset is the same at both sides and recorded with M851, the bed levelling should now work and not raise one side while squashing another.
 
[image]

shaddow501, 2019-12-07:
Dear David

I still do not believe that it is the issue but in order to test this assumption I have to try it. :)
I will try and report back.

have to wait another 5 hours till the printing will finish..

One other thing that could be related to the problem is the Z axis brass nuts that over time (& printing) wear out and can cause deviations.
[image]

dleask, 2019-12-07:
If that was the case with the brass nuts then you wouldnt get a clean print when you disable auto leveling...

shaddow501, 2019-12-07:
I do get a clean print though.

Attached please have a look how my BLTOUCH is installed.
[image]
[image]
[image]
[image]

dleask, 2019-12-07:
OK, let me be clear. The issue you are seeing is not related to wear on the Z-axis threaded rod or brass nuts.

shaddow501, 2019-12-07:
I have ordered so many parts for replacement but all are assumptions, nothing is certain.

dleask, 2019-12-07:
Can't help any further. Good luck

shaddow501, 2019-12-07:
Thanks by the way, if you replaced the TMC2208 also for the extruder motor didnt you experience issues with leaner advanced?
With my printer the extruder stop to extrude using ther TMC2208, with the normal driver it works.

shaddow501, 2019-12-07:
Dear dleask 

I followed your instructions and it is impossible to get the measurements between the bed and what the probe detect since there is an also a bed issue, if the bed is higher in one side no matter what I will do with the bars the height will remain.
But I have found out that I have a mechanical problem  that could relate to this issue.

On the right side of the printer when I try to lift that side with my hand I get about 2mm movement before the rod even start to turn. 

Maybe because of that the right side always get too low..
[image]

I have ordered those brass nuts, I will replace them to see if I get better results.

dleask, 2019-12-08:
If you have 2mm movement then something else is loose. The pitch of the threaded rod may give you 0.5mm when worn but 2mm is more than the pitch of the thread.

shaddow501, 2019-12-08:
Maybe I was exaggerating about the 2mm, with closer look it is more likely to be 1mm.
I have checked all the screws and all are tight.

shaddow501, 2019-12-09:
Dear  dleask

Can you please do me a favor do G29 when the bed is leveled and paste the results here, and then lift your right side of the bed (or with the bed screws or with some kind of 1mm paper, something that has a bit of height) only on the right side of the bed and again do G29 and paste the results?   

I want to be sure that the sensor I am using doesnt have a problem.
Also do you use the original bltouch or a clone?

Thanks.

dleask, 2019-12-09:
First output is soon as the bed hits 60C and hot end hits 200C. Bed is leveled with G29 then probed with G29.

Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 -0.070 +0.040 +0.143 +0.080 -0.030
Recv:  1 -0.118 +0.035 +0.117 +0.048 -0.082
Recv:  2 -0.085 -0.077 +0.115 +0.035 -0.070
Recv:  3 -0.280 -0.110 +0.090 +0.043 -0.055
Recv:  4 -0.145 -0.040 +0.138 +0.005 -0.058

[image]

Next, I put a steel rule laying across the Y axis on the right side of the bed.

Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 -0.095 -0.035 +0.120 +0.005 +0.952
Recv:  1 -0.248 -0.095 +0.107 +0.063 +0.985
Recv:  2 -0.182 +0.048 +0.155 +0.040 +0.995
Recv:  3 -0.190 +0.045 +0.128 +0.053 +1.030
Recv:  4 -0.090 +0.070 +0.148 +0.115 +1.057

[image]

Third run has the steel rule removed and rehomed and re leveled.

Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +0.038 +0.170 +0.250 +0.138 +0.020
Recv:  1 -0.010 +0.098 +0.247 +0.160 +0.005
Recv:  2 -0.020 +0.160 +0.222 +0.133 +0.010
Recv:  3 +0.027 +0.150 +0.230 +0.145 +0.012
Recv:  4 -0.025 +0.175 +0.225 +0.117 +0.022

[image]

Fourth run is rehomes and releveled as a consistency check against the readings from the third run.

Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 -0.025 +0.093 +0.190 +0.140 +0.050
Recv:  1 -0.077 +0.075 +0.175 +0.160 +0.015
Recv:  2 -0.072 +0.077 +0.212 +0.130 -0.002
Recv:  3 -0.063 +0.077 +0.190 +0.135 +0.022
Recv:  4 -0.063 +0.077 +0.268 +0.122 +0.030

[image]

The actual variation between readings (Set 3 - Set 4) is max 0.098 and min -0.043 which is near enough for what the probe is trying to achieve.

0 | 0.063 | 0.077 | 0.060 | -0.002 | -0.030
1 | 0.067 | 0.023 | 0.072 | 0.000 | -0.010
2 | 0.052 | 0.083 | 0.010 | 0.003 | 0.012
3 | 0.090 | 0.073 | 0.040 | 0.010 | -0.010
4 | 0.038 | 0.098 | -0.043 | -0.005 | -0.008

shaddow501, 2019-12-09:
Thanks, I will check on my side what do I get to see if the probe is working correctly.

by the way what did you use in order to see this graphic chart?

dleask, 2019-12-09:
It's OctoPrint running on a Raspberry Pi 3+ with the bed visualiser plugin.

shaddow501, 2019-12-09:
Here are some pictures.

1st on is with running when bed temp is around 70C when the bed is some how leveled with the normal operation (without using the bed leveling feature)
Recv:       0      1      2      3      4
Recv:  0 +0.442 +0.435 +0.397 +0.355 +0.452
Recv:  1 +0.402 +0.460 +0.502 +0.492 +0.452
Recv:  2 +0.410 +0.452 +0.502 +0.487 +0.457
Recv:  3 +0.422 +0.480 +0.475 +0.500 +0.457
Recv:  4 +0.490 +0.440 +0.497 +0.532 +0.455
[image]

2nd run is when I put envelope on the right side of the bed.
Recv:       0      1      2      3      4
Recv:  0 +0.437 +0.447 +0.512 +1.222 +1.310
Recv:  1 +0.417 +0.395 +0.387 +1.467 +1.187
Recv:  2 +0.390 +0.427 +0.402 +1.995 +1.095
Recv:  3 +0.422 +0.457 +0.495 +1.785 +1.090
Recv:  4 +0.482 +0.520 +0.515 +2.202 +1.455

[image]

3rd run is when I remove the envelop and run G29 again.
Recv:       0      1      2      3      4
Recv:  0 +0.440 +0.447 +0.512 +0.510 +0.457
Recv:  1 +0.400 +0.402 +0.510 +0.462 +0.455
Recv:  2 +0.405 +0.427 +0.392 +0.502 +0.472
Recv:  3 +0.417 +0.462 +0.427 +0.470 +0.445
Recv:  4 +0.470 +0.457 +0.530 +0.367 +0.490
[image]

As you can see 1st run and 3rd run the values are pretty much close, you can see that the right side is a bit higher than the left, so why when printing with bed leveling function on the operation is opposite it raise the nozzle on the left side (which should be lower) and lower the nozzle on the right side (which it should raise it)....

dleask, 2019-12-09:
Firstly your initial levelling is not right as all readings are at least 0.35mm too high. Assuming you have levelled each corner with a piece of paper no thicker than 0.1mm then your Z offset is out by 0.3mm. I suspect it should be set to M851 Z-2.4.

Secondly, if the right side is still too tight it confirms what I originally said that your X axis rails are out of alignment on the right side and you need to loosen the top rail on the right and move it back then tighten up the rail mount.

shaddow501, 2019-12-09:
>  Firstly your initial levelling is not right as all readings are at least 0.35mm too high. Assuming you have levelled each corner with a piece of paper no thicker than 0.1mm then your Z offset is out by 0.3mm. I suspect it should be set to M851 Z-2.4.

This is not correct since the of the installation of the bltouch location and height.
for me to be scratching the bed with a paper it is +-0.4 for you since your bltouch is lower than mine it is closer to 0.

Check yourself what you are saying is not correct.
Again the value is according to the bltouch installation height.

> Secondly, if the right side is still too tight it confirms what I originally said that your X axis rails are out of alignment on the right side and you need to loosen the top rail on the right and move it back then tighten up the rail mount.

Your assumption could have been right if it was too tight also on the left side but on the left side it is too high and the probe detects that the bed is lower from the other side, I am sorry but I cannot find any sense in what you are saying.

If the middle of the bed is ok, so according to what you say I should have the same results with the left side of the bed that should be close as like the middle of the bed, but instead the nozzle goes higher. much higher than the middle height.

I get very nice prints about the same height in both of the side of the bed when not using auto bed leveling
Only with bed leveling I get worse performance.  

And another thing, it shows me that the probe is getting the correct values just it seems that the left and right are mixed.

dleask, 2019-12-10:
I can’t help you any further. I suggest you continue without the BLTouch if that gives you good prints. 

Bye

benzoll, 2020-08-07:
Hello
I have the same problem as shaddow501
Has a solution been found to the problem

BigFudgeDD, 2021-04-18:
@dleask : Hope you read this almost one and a half year later. I just registered at github for the sole reason to give you a big THANK YOU for you hint. I had the same issue, that after installing a BLTouch and printing a bed level test, the squares at the left where to close, perfect in the middle and to far away on the right side. I searched the error for three days and tried three different Marlin builds, always the same problem. It drove me nuts.

I don't know, if I would have guessed the issue with the twisted rods on my own, but after your hint it was obvious immediately. I measured a deviation in the Z offset between left and right side of 0.19 mm, so the rods had to be off by 0.38 mm. After three or four iterations of readjusting the rods, I was down to 0.04 mm deviation in the Z offset. Still not perfect, but all four ends of my rods are at their mechanical limit, so no more room for further improvement. But it does the trick now and the ABL works just fine.

So, again thank you very much for pointing me in the right direction and happy greetings from Germany.