MNieddu91/Marlin-AI3M-BLTouch #20: [BUG] BLTouch not deploying

Psychokiller1888, 2019-12-06:
**Describe the bug**
Flashed the latest sources after compiling via Arduino IDE. Had to reverse axis. Homing works fine. BLTouch 3.1 does self check and passes it, stays solid red after printer power on, with pin retracted. G29 starts the leveling cycle but the pin is never deployed, resulting in printer to stop by hitting Z endstops

**To Reproduce**
Steps to reproduce the behavior:
1. Start printer
2. G28
3. G29
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Photos of the print / output of the terminal**
If applicable, add photos or console outputs to help explain your problem.

**Printer (please complete the following information):**
 - Firmware version: Cloned repo, compiled master
 - Through either Cura terminal or SD card bed leveling special menu item
 - i3 Mega, Mega-S

Psychokiller1888, 2019-12-07:
Found only now some explanation on your wiki, sorry for the disturbance

Psychokiller1888, 2019-12-07:
Nop, even after firmware adaptation, it won't deploy. Tried to reverse the black and white pins also

Rich990, 2019-12-07:
Have you Adapter the version of your trigorilla board? That did it in my case.

Psychokiller1888, 2019-12-07:
Nope, I did not. What do you mean by that?

Rich990, 2019-12-07:
Somewhere in the firmware (i don’t know the exact line) it says something like „for trigorilla version xx put a ‚0‘ for version yy put ‚1‘....

Rich990, 2019-12-07:
Line 1964-> if you have trigorilla 1.1 then Set a 1, otherwise 0.

Psychokiller1888, 2019-12-07:
This did it, many thanks !!

closed 2019-12-07 by Psychokiller1888