MNieddu91/Marlin-AI3M-BLTouch #21: Problem with Original BLTouch v3.1 in Marlin 1.1.9

RedJohn14, 2019-12-31:
Hello,

**I installed Marlin Build from your repo here: (v.1.1.9)**
https://github.com/MNieddu91/Marlin-AI3M-BLTouch

**My settings in configuration.h looks like this:**

**I used the original BLTouch v3.1 from here:**
https://www.amazon.de/gp/product/B075B57CHD/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

**I also tested your new repo Marlin 2.0.x:**
https://github.com/MNieddu91/Marlin-Ai3M-2.0.x-BLTouch

I have home everything with G28 and use G29 all works perfekt. BLTouch Pin comes out but if I TRIGGERED it from hand nothing happens Z axis not stop!

If I home everything again with G28 and looks with M119 I see this:
< 13:10:36: x_min: TRIGGERED
< 13:10:36: y_min: TRIGGERED
< 13:10:36: z_min: TRIGGERED
< 13:10:36: z2_min: TRIGGERED
< 13:10:36: z_probe: open

The z_probe status won't change. I TRIGGERED the Pin manual but the status won't change. The Server cable on my motherboard is correct because I can use this command without any problem:

**M280 P0 S10 ; pushes the pin down
M280 P0 S90 ; pulls the pin up
M280 P0 S120 ; Self test – keeps going until you do pin up/down or release alarm
M280 P0 S160 ; Release alarm**
 
Only the trigger status will not work Z axis not stop! Any idea what the problem is?

Attached you find a picture from my BLTouch mount in the printers head. And a picture from the cable on my motherboard. I have the new Board 1.1.

[image]

[image]

lyndsysimon, 2020-02-16:
I had the same problem, but believe I have resolved it. Following the instructions in the `Configuration.h` file, I swapped the black and white wires. That allowed me to stop the Z-movement by manually triggering the pin.

I'm continuing to work out my configuration now, and will update here if I have any further issues.

lyndsysimon, 2020-02-16:
All seems well now - I've set my Z-offset and it seems to level correctly. Pending a test print, I'm calling this a success, and will set about cleaning up my wire mess :)

closed 2020-04-07 by MNieddu91