MNieddu91/Marlin-AI3M-BLTouch #26: Bltouch antclabs v3.1 probing failure

saboaua, 2020-07-29:
I flash my anycubic with the firmware but cannot get the bltouch probing. Octoprint gives me the error message probing failure.

I have enabled the only the define bltouch v3 and luck. I have switch the white and black and still no luck with the bltouch.

What could be the issue?

yzheka-m, 2020-10-22:
I have the same issue, @saboaua Any progress on this?

wes1993, 2021-01-30:
@yzheka-m have you found a solution or another firmware?

saboaua, 2021-10-31:
No, didn't solve the issue and using the manual process

phyex0, 2021-10-31:
> No, didn't solve the issue and using the manual process

Damn.. it's killing me

jj-uk, 2022-01-02:
I'm using Marlin 2.0.7.2 (https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.7.2).
My BLTouch is a 3.1.
My board is a Trigorilla 1.1. 

Attached are my config files for you to try. Place them into \Marlin-2.0.7.2\Marlin\ folder and re-compile Marlin.

In Configuration.h, you MUST check the values of line 147 (BOARD_TRIGORILLA_14_11) and line 1024 (NOZZLE_TO_PROBE_OFFSET). 
Pay special attention to the Z-value here, mine is -3.9. I'd suggest you set this to zero to start with, then follow the process here: https://github.com/MNieddu91/Marlin-AI3M-BLTouch/wiki/Adjust-configuration - starting from "Next we determine the value for the Z-Offset".

All changes I've made in the config file are marked with "JXB". Search for "JXB" in the config files to find my changes, if required. I've added comments explaining the change and what the default value was previously.

I have set the probing to be 3x each point, 8x8 points. It takes a while but it very accurate.

I perform a bed level every few months using optoprint. Here is the octoprint code for the "Bed Visualizer" plugin (https://plugins.octoprint.org/plugins/bedlevelvisualizer/) :
`M280 P0 S90                            ; retract pin
M155 S30                                 ; reduce temperature reporting rate to reduce output pollution
M140 S60                                 ; starting by heating the bed for nominal mesh accuracy
G28                                           ; home all axes
M190 S60                                 ; waiting until the bed is fully warmed up
G28                                          ; home all axes
@BEDLEVELVISUALIZER      ; tell the plugin to watch for reported mesh
G29 T                                       ; run bilinear probing
M155 S3                                  ; reset temperature reporting
M140 S0                                  ; cooling down the bed
M280 P0 S90                          ; retract pin
G28                                         ; home all axis
M500                                       ; save results to eeprom`

And finally, here is my Cura printer "Start G-Code". Required so that it uses the stored bed level info for each print:
`
; Script based on an original created by tjjfvi (https://github.com/tjjfvi)
; An up-to-date version of the tjjfvi's original script can be found
; here:  https://csi.t6.fyi/
; Note - This script will only work in Cura V4.2 and above!
; --- Global Settings
; layer_height = {layer_height}
; smooth_spiralized_contours = {smooth_spiralized_contours}
; magic_mesh_surface_mode = {magic_mesh_surface_mode}
; machine_extruder_count = {machine_extruder_count}
; --- Single Extruder Settings
; speed_z_hop = {speed_z_hop}
; retraction_amount = {retraction_amount}
; retraction_hop = {retraction_hop}
; retraction_hop_enabled = {retraction_hop_enabled}
; retraction_enable = {retraction_enable}
; retraction_speed = {retraction_speed}
; retraction_retract_speed = {retraction_retract_speed}
; retraction_prime_speed = {retraction_prime_speed}
; speed_travel = {speed_travel}

G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
M501 ; restore settings
M420 S1 ;use bed levelling data
G1 Z15.0 F{speed_travel} ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
G1 F{speed_travel}
G0 Y20 F{speed_travel}
M117 Printing...
G5`

And here is my firmware version:
`Send: M115
Recv: FIRMWARE_NAME:Marlin 2.0.7.2 (Jan 20 202
[Marlin.zip](https://github.com/MNieddu91/Marlin-AI3M-BLTouch/files/7799596/Marlin.zip)
1 22:04:59) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Anycubic i3 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff`